MCUdude/MegaCoreX #193: Error uploading bootloader using PlatformIO, SerialUPDI

rebebba, 2024-02-14:
We are trying to burn a bootloader to a custom ATMega4809 board over serialUPDI. We are following the [.ini template provided](https://github.com/MCUdude/MegaCoreX/blob/master/PlatformIO.md#platformioini-templates), with adjustments to fit our board specifications

```
board_build.f_cpu = 4000000L
board_build.variant = 40pin-standard
upload_port = COM3
board_hardware.uart = uart0
```

Running code

`pio run -e fuses_bootloader -t bootloader -v`

Produces the error

```
avrdude: 1 byte of lock verified
avrdude: leaving NVM programming mode
avrdude serialupdi_close() warning: releasing DTR/RTS handshake lines

avrdude done.  Thank you.

avrdude -p atmega4809 -C C:\Users\rn71\.platformio\packages\tool-avrdude\avrdude.conf -P usb -c serialupdi -xrtsdtr=high -U flash:w:C:\Users\rn71\.platformio\packages\framework-arduino-megaavr-megacorex\bootloaders\optiboot\bootloaders\mega0\115200\Optiboot_mega0_UART0_DEF_115200_A7.hex:i
avrdude error: cannot open port usb: The system cannot find the file specified.

avrdude error: unable to open programmer serialupdi on port usb

avrdude done.  Thank you.

*** [bootloader] Error 1
================================================= [FAILED] Took 3.65 seconds ===========================================

Upload_UPDI       IGNORED
Upload_UART       IGNORED
fuses_bootloader  FAILED    00:00:03.647
============================================ 1 failed, 0 succeeded in 00:00:03.647 ============================================
```

We have the same issue trying to burn a bootloader to the ATMega4808 board. We have no problem uploading to the board if a bootloader is already burned (done in Arduino IDE).
Any idea what we may be doing wrong? Thanks in advance!

MCUdude, 2024-02-14:
The problem is `-P usb` in the avrdude command. Can you please share your entire platformio.ini, so I can try to reproduce the issue?

rebebba, 2024-02-14:
Hello! Yes, of coarse. Please see below:

```
[env]
platform = atmelmegaavr
framework = arduino

board = ATmega4809
; Clock frequency in [Hz]
board_build.f_cpu = 4000000L
; Oscillator type (internal or external)
board_hardware.oscillator = internal
; Arduino pinout variant
board_build.variant = 40pin-standard

; Monitor port is auto detected. Override here
;monitor_port =
monitor_speed = 9600
monitor_dtr = 0

; Run the following command to upload with this environment
; pio run -t upload
[env:Upload_UPDI]
; Upload protocol for UPDI upload
upload_protocol = serialupdi
; COM1 or COM3
upload_port = COM3
upload_speed = 115200
upload_flags =
  -xrtsdtr=high

; Run the following command to upload with this environment
; pio run -e Upload_UART -t upload
[env:Upload_UART]
; Upload protocol for serial uploads (using Optiboot)
upload_protocol = arduino
; COM1 or COM3
upload_port = COM3
upload_flags =

; run the following command to set fuses
; pio run -e fuses_bootloader -t fuses
; run the following command to set fuses + burn bootloader
; pio run -e fuses_bootloader -t bootloader -F -v
[env:fuses_bootloader]
; Inherit upload settings from the Upload_UPDI environment
extends = env:Upload_UPDI
; Hardware settings
board_hardware.bod = 2.6v
board_hardware.eesave = yes
board_hardware.uart = uart0
board_hardware.rstpin = reset
```

MCUdude, 2024-02-16:
Sorry for the late reply. I've been quite busy the recent days, and haven't had time to look at this issue until now.
I can reproduce your issue, and it's clearly a bug in the PlatformIO system.

I'm the one who designed the fuses/bootloader logic in the MegaCoreX PlatformIO, so I'll look into it and provide a PR that hopefully can fix this issue. Stay tuned!

MCUdude, 2024-02-16:
I found the root cause! See PR here: https://github.com/platformio/platform-atmelmegaavr/pull/67

rebebba, 2024-02-16:
> I found the root cause! See PR here: platformio/platform-atmelmegaavr#67

Thank you!

Rebecca Nagel
***@***.***

closed 2024-02-23 by rebebba