MCUdude/MegaCoreX #195: Addition of new programming adapter (Nano Every)

ltwin8, 2024-03-06:
Hello, I would like to ask if it is possible to officially support the bare Arduino nano every (atmega4809 removed) as a programming adapter?

the nano every itself is already supported and does only require a modified boards.txt and programmer.txt to work. I used it successfully with the 4809 on target PCB and 809 on target PCB.

The reason behind this request is the cheap and easy way to get hands on a capable Programmer which includes a convenient serial adapter as well while also being able to supply 1.2A of current to the DUT.

I have multiple nano every boards without the 4809 (chip shortage, needed to salvage them sadly) and I would like to use them as programmers in the official core without every time having to fiddle with files which get overwritten when I update Arduino.

i would also be happy to give instruction and casing for a nano every to program adapter conversion.

thanks for any help.

MCUdude, 2024-03-06:
> Hello, I would like to ask if it is possible to officially support the bare Arduino nano every (atmega4809 removed) as a programming adapter?

That was the original idea, but the current solution is like this for a reason.
The Nano Every programmer chip serves a dual purpose programmer and USB to serial adapter.
By default, it's in USB to serial mode, but when Arduino IDE sends a 1200bps touch (connecting with a baud rate of 1200 baud), the programmer chip enters programming mode and continues when Arduino IDE calls Avrdude.

The problem is that the 1200bps thing can't be issued when using the Nano Every as a dedicated programmer from programmers.txt. It's an Arduino IDE limitation. There is a workaround in the latest Avrdude 7.3 version, but this is not yet available for bundling with 3rd party Arduino cores, as there are no statically linked binaries available yet.

closed 2024-03-06 by MCUdude

ltwin8, 2024-03-06:
I have modified the boards and programmers and what should I say it works, I added a new section in boards.txt 

i will attach file later.

it’s like a copy of nano every, named it nano every on PCB, gave it 48pin standard, and same options as the nano every.

than took this block, copying and deleted the „4“ of „4809.xyz.(…)“ and put this section below the last 809 entry.

this way I have successfully programmed 4809 and 809 while also being able of serial comm

MCUdude, 2024-03-06:
Well, if you got it working, feel free to submit a PR! The bug/limitation may have been fixed since I first did the Nano Every implementation.

The plan was to look at the issue again when I could bundle suitable Avrdude 7.3 binaries, but if you already have a solution, that would be just as good.

ltwin8, 2024-03-06:
```
#for 4809
4809.menu.pinout.nano_every_on_PCB=Nano Every on PCB
4809.menu.pinout.nano_every_on_PCB.build.variant=48pin-standard
4809.menu.pinout.nano_every_on_PCB.upload.tool=avrdude_nanoevery
4809.menu.pinout.nano_every_on_PCB.upload.use_1200bps_touch=true
4809.menu.pinout.nano_every_on_PCB.upload.protocol=jtag2updi
4809.menu.pinout.nano_every_on_PCB.program.extra_params=-P{serial.port} -e
4809.menu.pinout.nano_every_on_PCB.build.compat=

#for 809
809.menu.pinout.nano_every_on_PCB=Nano Every on PCB
809.menu.pinout.nano_every_on_PCB.build.variant=48pin-standard
809.menu.pinout.nano_every_on_PCB.upload.tool=avrdude_nanoevery
809.menu.pinout.nano_every_on_PCB.upload.use_1200bps_touch=true
809.menu.pinout.nano_every_on_PCB.upload.protocol=jtag2updi
809.menu.pinout.nano_every_on_PCB.program.extra_params=-P{serial.port} -e
809.menu.pinout.nano_every_on_PCB.build.compat=
```

works without issues...

i do not exactly understand why this is not possible to add since this option is usable

MCUdude, 2024-03-06:
That will probably work, but it's a hack for sure. You'll need a "Nano Every on PCB" option for every pinout. It would be much better to have a dedicated programmer option in programmers.txt that says "Nano Every as programmer". However, this will be possible when using Avrdude 7.3.

MCUdude, 2024-03-06:
> However is a pull possible if submitted or won’t you accept this „hack“?

Sorry, but I don't think I'll accept such a PR. I'd rather wait until I can bundle Avrdude 7.3.

ltwin8, 2024-03-06:
Okay, thanks for the answer, no problem then I will host locally.

i was thinking maybe it would be a positive addition for other users too.

however I hope the people interested can replicate it themselves.

per1234, 2024-03-06:
FYI, Arduino's name for this programmer is "MuxTO":

https://github.com/arduino/ArduinoCore-megaavr/blob/master/firmwares/MuxTO/MuxTO.ino#L12

> Serial-to-USB converter + UPDI programmer on the Arduino Nano Every (MuxTO for brevity)

Since the "MuxTO" firmware is open source (and even relatively easy to modify, compile, and upload since it is an Arduino sketch), and the ATSAMD11 is available in a hand soldering-friendly package, it is possible that the community members might create a "MuxTO" programmer from scratch rather than adapting a Nano Every board (though the price of the Nano Every is low enough that buying one for the MuxTO alone is reasonable).

ltwin8, 2024-03-06:
The package is no problem, the issue was only related if the nano every PCB would be officially supported as a programmer since they are a cheap and easy way of getting a UPDI adapter + Serial + 5V at >1A from 7-20V input

ltwin8, 2024-03-06:
> @ltwin8 you failed to understand the purpose of my comment. It wasn't directed at you specifically anyway so I'll move on to more productive endeavors.
> 
> Best wishes.

Okay, harsh reply but okay…
Anyway thanks for the input.