MCUdude/MegaCoreX #200: avrdude ur_initstruct() error: bootloader might be optiboot 6.2? Please use -xbootsize=<num>

ednieuw, 2024-09-17:
When uploading to ATMEGA 1284P compiled with Arduino IDE 1 and IDE 2  I got his upload error.
I use an FTDI connected to an PCB where the 1284P is on
I have four PCB boards that uploaden fine in 2019-2020 but none at the moment. 

I downgraded  from 1.1.2 to 1.0.10 but got the same error.

Any idea what to do?

```

Sketch uses 30758 bytes (23%) of program storage space. Maximum is 130560 bytes.
Global variables use 3193 bytes (19%) of dynamic memory, leaving 13191 bytes for local variables. Maximum is 16384 bytes.
C:\Users\ednie\AppData\Local\Arduino15\packages\MightyCore\tools\avrdude\7.2-arduino.1/bin/avrdude -CC:\Users\ednie\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\3.0.2/avrdude.conf -v -V -patmega1284p -curclock -PCOM19 -b115200 -D -xnometadata -Ueeprom:w:C:\Users\ednie\AppData\Local\Temp\arduino_build_219378/Character_Colour_Clock_V090.ino.eep:i -Uflash:w:C:\Users\ednie\AppData\Local\Temp\arduino_build_219378/Character_Colour_Clock_V090.ino.hex:i 

avrdude: Version 7.2-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\ednie\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\3.0.2\avrdude.conf

         Using Port                    : COM19
         Using Programmer              : urclock
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega1284P
         Chip Erase delay              : 55000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash                  65    10   128    0 yes    131072  256    512  4500  4500 0x00 0x00
           lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : Urclock
         Description     : Urclock programmer for urboot bootloaders using urprotocol
         Bootloader using STK500v1 communication protocol
avrdude: AVR device initialized and ready to accept instructions
avrdude ur_initstruct() error: bootloader might be optiboot 6.2? Please use -xbootsize=<num>

avrdude avr_signature() error: unable to read signature data for part ATmega1284P, rc=-1
avrdude main() error: unable to read signature data, rc=-1

avrdude done.  Thank you.

the selected serial port 
 does not exist or your board is not connected

```

MCUdude, 2024-09-17:
You should probably update the bootloader that's on your board to Urboot, the bootloader MightyCore currently uses. If that's not an option, you'll have to downgrade to MightyCore v2.x

ednieuw, 2024-09-18:
A downgrade to MightyCore v2.x was the solution. I just had to correct a few lines of code. 
Thanks for the quick reply.

ednieuw, 2024-09-18:
Problem solved

closed 2024-09-18 by ednieuw