MCUdude/MegaCoreX #201: I can not compile Ethernet.h on last - MegaCoreX v1.1.3

mgy48, 2024-11-17:
My arduino do not compile when included Ethernet.h library with MegaCoreX v1.1.3
There is no problem with old versions.
Please check it. Do i have a mistake?

MCUdude, 2024-11-17:
Can you post the error you're getting?

mgy48, 2024-11-17:
MegaCoreX v1.1.3

[image]

There is any error with old verions.

MCUdude, 2024-11-17:
This is hardly helpful at all. You'll have to copy the entire output from the Arduino IDE.

mgy48, 2024-11-17:
My test code is:

```
#include <Ethernet.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}

```

the entire output is:

```
FQBN: MegaCoreX:megaavr:4808
Using board '4808' from platform in folder: C:\Users\user\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.3
Using core 'coreX-corefiles' from platform in folder: C:\Users\user\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.3

Detecting libraries used...
C:\Users\user\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++17 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=atmega4808 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_ATmega4808 -DARDUINO_ARCH_MEGAAVR -IC:\Users\user\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.3\cores\coreX-corefiles/api/deprecated -IC:\Users\user\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.3\cores\coreX-corefiles -IC:\Users\user\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.3\variants\32pin-standard C:\Users\user\AppData\Local\Temp\arduino\sketches\62E2E3EFDD73DEE05A85442B7B18B5D6\sketch\sketch_nov17a.ino.cpp -o nul
Alternatives for Ethernet.h: [Ethernet@2.0.1 Ethernet@2.0.2]
ResolveLibrary(Ethernet.h)
  -> candidates: [Ethernet@2.0.1 Ethernet@2.0.2]
C:\Users\user\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++17 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=atmega4808 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_ATmega4808 -DARDUINO_ARCH_MEGAAVR -IC:\Users\user\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.3\cores\coreX-corefiles/api/deprecated -IC:\Users\user\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.3\cores\coreX-corefiles -IC:\Users\user\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.3\variants\32pin-standard -Ic:\Users\user\Documents\Arduino\libraries\Ethernet\src C:\Users\user\AppData\Local\Temp\arduino\sketches\62E2E3EFDD73DEE05A85442B7B18B5D6\sketch\sketch_nov17a.ino.cpp -o nul
C:\Users\user\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++17 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=atmega4808 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_ATmega4808 -DARDUINO_ARCH_MEGAAVR -IC:\Users\user\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.3\cores\coreX-corefiles/api/deprecated -IC:\Users\user\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.3\cores\coreX-corefiles -IC:\Users\user\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.3\variants\32pin-standard -Ic:\Users\user\Documents\Arduino\libraries\Ethernet\src c:\Users\user\Documents\Arduino\libraries\Ethernet\src\Dhcp.cpp -o nul
Alternatives for SPI.h: [SPI@1.0]
ResolveLibrary(SPI.h)
  -> candidates: [SPI@1.0]
C:\Users\user\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++17 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=atmega4808 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_ATmega4808 -DARDUINO_ARCH_MEGAAVR -IC:\Users\user\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.3\cores\coreX-corefiles/api/deprecated -IC:\Users\user\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.3\cores\coreX-corefiles -IC:\Users\user\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.3\variants\32pin-standard -Ic:\Users\user\Documents\Arduino\libraries\Ethernet\src -IC:\Users\user\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.3\libraries\SPI\src c:\Users\user\Documents\Arduino\libraries\Ethernet\src\Dhcp.cpp -o nul
C:\Users\user\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++17 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=atmega4808 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_ATmega4808 -DARDUINO_ARCH_MEGAAVR -IC:\Users\user\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.3\cores\coreX-corefiles/api/deprecated -IC:\Users\user\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.3\cores\coreX-corefiles -IC:\Users\user\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.3\variants\32pin-standard -Ic:\Users\user\Documents\Arduino\libraries\Ethernet\src -IC:\Users\user\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.3\libraries\SPI\src c:\Users\user\Documents\Arduino\libraries\Ethernet\src\Dns.cpp -o nul
C:\Users\user\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++17 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=atmega4808 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_ATmega4808 -DARDUINO_ARCH_MEGA [... cut ...]
   uint8_t* raw_address() { return _address.bytes; };
            ^~~~~~~~~~~
Multiple libraries were found for "Ethernet.h"
  Used: C:\Users\user\Documents\Arduino\libraries\Ethernet
  Not used: C:\Users\user\AppData\Local\Arduino15\libraries\Ethernet
Using library Ethernet at version 2.0.2 in folder: C:\Users\user\Documents\Arduino\libraries\Ethernet 
Using library SPI at version 1.0 in folder: C:\Users\user\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.3\libraries\SPI 
exit status 1

Compilation error: exit status 1
```

MCUdude, 2024-11-17:
Thanks! A recently merged PR, #198 is what's causing this. I'll look into what to do in order to fix this

PR #202 (2024-11-20): Fix IPAddress further so that the Ethernet library will work again
This should fix #201 
This is my fault, I applied the minimal number of changes necessary to get ArduinoModBus to compile. Apparently, in the ArudinoCore-API they have to forward declare several classes outside the arduino namespace to allow EthernetClass, DhcpClass, and DNSClient to be friends. 

I have already tested this change manually using the TelentClient example and compilation is once again successful.
closed 2024-11-20 by MCUdude