MCUdude/MegaCoreX #204: Serial comunication problem - Fingerprint Scanner - TTL (GT-521F52)

revizak, 2025-01-13:
HW  Nano EVERY 4808 
I have a problem with GT-521F52 FPS module. The module does not respond at all to a query via sotwareserial.
I also tested SIM800L , that one goes fine.
When I connect GT-521F52 module to arduino uno/nano with ATMEGA328P it also goes without problems.
I then connected a logic analyzer to the serialport and compared the data with the ATMEGA328p and 4808.

more in the picture the top part is 4808 and the bottom part is 328
[image]

Is it possible to fix this problem or is it a bug in the sofwareserial library?

MCUdude, 2025-01-15:
Can you try to replace the MegaCoreX SoftwareSerial library with the one provided by Arduino themselves?

https://github.com/MCUdude/MegaCoreX/tree/master/megaavr/libraries/SoftwareSerial

revizak, 2025-01-15:
Today I used to test the original arduino NANO EVERY with 4809 and their library, the result is the same, the data on the logic analyzer show the same problem, so the library of the company ARDUINO develops the same problem of eight cycles of logic zero instead of one start bit, I also contacted the technical support of ARDUINO

revizak, 2025-01-16:
today I tried to replace softwareserial library, so it didn't even compile

cattledogGH, 2025-01-17:
Along with the uart connected to usb, there are three additional hardware serial ports available on the Nano Every.  Why do you want to use Software Serial on this platform?

revizak, 2025-01-17:
I don't know what all needs to be modified to make the Fingerprint_Scanner-TTL library use the HW serial port . In the setup only FPS_GT511C3 fps(4, 5); // (Arduino SS_RX = pin 4, Arduino SS_TX = pin 5) but in the **FPS_GT511C3.h** file there is a rank `#include "SoftwareSerial.h"` so probably more modifications are needed for the HW serial communication

cattledogGH, 2025-01-17:
There is a fork of the GT511C3 library written for the Arduino Mega and Due, using hardware Serial1. 

https://github.com/ifsc-saojose/FPS_GT511C3_DUE_MEGA/tree/master

It is not available using the Arduino Library manager, but can be downloaded and installed with the zip file.

Seeing the example code, you should be able to run that on the Nano Every with this constructor.

//FPS_GT511C3 fps(19, 18); //Mega and Due Serial1  rx,tx pins

FPS_GT511C3 fps(0, 1); //Nano Every Serial1 rx,tx pins

I see that the fingerprint scanner UART logic level is 3.3V. You will need a logic level converter or voltage divider to safely communicate with a 5V device like the Nano Every.

[image]

revizak, 2025-01-18:
Thank you cattledogGH, this could be a great solution , but I use the GT-521F52 module, which is for 3000 fingerprints, and the library from ifsc-saojose is for the 200 fingerprint version. So also here the library needs to be modified.

cattledogGH, 2025-01-18:
>  but I use the GT-521F52 module, which is for 3000 fingerprints

Then you are willing to modify the library after all.
From  [https://learn.sparkfun.com/tutorials/fingerprint-scanner-gt-521fxx-hookup-guide/al](url)

> Adjust Code for GT-521F52's 3000 ID's
Depending on what model you are using, make sure to change number of fingerprint IDs. For example,the GT521F52 can hold 3000 fingerprint IDs instead of 200. When enrolling a fingerprint to the GT-521F52, you will need to make sure to change retval < 200 to retval < 3000 for the FPS_GT511C3::Enroll1(), FPS_GT511C3::Enroll2(), and FPS_GT511C3::Enroll3() functions in the [FPS_GT511C3.cpp](https://github.com/sparkfun/Fingerprint_Scanner-TTL/blob/master/src/FPS_GT511C3.cpp) file.

When identifying a fingerprint with the GT-521F52, make sure to update if (retval > 200) retval = 200; to if (retval > 3000) retval = 3000; for the FPS_GT511C3::Identify1_N() function in the FPS_GT511C3.cpp file. You will also need change id <200 to id <3000 in the [FPS_IDFinger.ino](https://github.com/sparkfun/Fingerprint_Scanner-TTL/blob/master/examples/FPS_IDFinger/FPS_IDFinger.ino) file.

Try testing the scanner with the code to see if the scanner is able to read the fingerprints that were enrolled.

You should be able to make the same changes to the ifsc-saojose library written for Serial1.

revizak, 2025-01-19:
Thank you cattledogGH, I'm going to test it and let you know what the result is. For me this is probably the solution but it doesn't solve the problem of eight bit cycles before the start bit when using the softwareserial library.

revizak, 2025-01-26:
As ARDUINO EVERY is a disappointment for me , MEGA 2560 PRO is quite big and expensive , so I tested SIM800L , GT-521F52 and display SH1106 on RP2040 Zero and everything goes without problems. I decided to put all my equipment on the RP2040.

[image]

 Anyway, I would be happy if the problem of serial communication on ARDUINO EVERY could be eliminated

cattledogGH, 2025-01-26:
>I would be happy if the problem of serial communication on ARDUINO EVERY could be eliminated 

Did you ever hear back from Arduino customer support?

Do you see the 8 cycles of logic 0 instead of 1 start bit with Serial1 as well as with software serial?

It's interesting that other devices you have tested (Sim800l) work ok with the signals from the Nano Every and there is a specific issue with the GT-521F52.

revizak, 2025-01-26:
Arduino customer support with no response. Interestingly, I didn't even get a notification when I sent the problem to support, I tried it twice.

The 8 start cycles are on the softwareserial, when I used the HW serial there were nonsense on the analyzer. I can still connect the SIM800 to the 4808 and 4809 and see with the analyzer what data they are sending. This will check if this is a specific GT-521F52 problem or not.

revizak, 2025-02-01:
So today I took some time and looked at the data from the 328 vs. 4808 when communicating with the SIM800 via SoftwareSerial.

328
[image]

4808
[image]

revizak, 2025-02-01:
SIM800 communicates using AT commands and when it does not receive a response the query goes again. With GT-521F52 the HEX string is sent with ARDUINO and when it does not receive a reply the search stops. Anyway, it is probably a problem of the SoftwareSerial library that it sends the first data from logical 0.

revizak, 2025-02-02:
4808 HardwareSerrial is OK
[image]

cattledogGH, 2025-02-02:
> 4808 HardwareSerial is OK

That's as expected and good news. There is absolutely no need to use Software Serial with the Nano Every as there are three additional hardware serial ports available with MegaCoreX.

revizak, 2025-02-03:
OK,it can be closed

closed 2025-02-04 by revizak