MCUdude/MegaCoreX #210: Programming the Nano Every using 'Upload using Programmer'

felias-fogg, 2025-08-06:
I am puzzled that you cannot program the Nano Every by selecting JTAG2UPDI as the programmer, choosing the right port, and then pressing ⌘-Shift-U (`Upload using Programmer`). The only difference to a simple `Upload` is that the '-e' option is not present when calling avrdude.

MCUdude, 2025-08-06:
The Nano Every is implemented in a slightly different way, because it requires a 1200bps "touch" to enter programming mode. The Avrdude version that's bundled with MegaCoreX is built without [libserialport](https://sigrok.org/wiki/Libserialport), which means that extended serial port functionality such as `-r` (1200bps touch using Avrdude)  doesn't work.

It is possible to trigger the Arduino IDE 1200bps touch using boards.txt, but it cannot be done using programmers.txt. It has nothing to do with the `-e` flag.

https://github.com/MCUdude/MegaCoreX/blob/450c39e4c18c518284a21566a24db33b2cf21c5c/megaavr/boards.txt#L44-L50

https://github.com/MCUdude/MegaCoreX/blob/450c39e4c18c518284a21566a24db33b2cf21c5c/megaavr/programmers.txt#L33-L37

per1234, 2025-08-07:
It is exposed as a test pad on the bottom of the board:

[image]

<sub>

<details markdown="1">
  <summary>📷</summary>
  <p markdown="block">

[ABX00028-full-pinout.pdf](https://docs.arduino.cc/resources/pinouts/ABX00028-full-pinout.pdf) by [Arduino](https://www.arduino.cc/) - [CC BY-SA 4.0](https://creativecommons.org/licenses/by-sa/4.0/) (cropped, border added)

  </p>
</details>

</sub>

felias-fogg, 2026-04-19:
@per1234 @MCUdude Works perfectly! All tests of my test battery have been passed.

[image]