MCUdude/MegaCoreX #211: register number above 15 required

ZoharML, 2025-08-09:
I have a big code for my AtMega4808.
From time to time I got this annoying error. but was able to make some minimal "random" change in my code and over come this.
But after my last changes to the code I'm just stuck with this error.

I tryied to remove the LTO from "platform.txt" but then I was not even able to compile the empty code created after pressing "new sketch" in arduino IDE.

What can be done about this?
My code is very large and proprietry, I cannot share it publicly :( 
Can I make some compiler report that would help find the source of this?

```
d:\temp\wintemp\ccihhjdj.s: Assembler messages:
d:\temp\wintemp\ccihhjdj.s:10006: Error: register number above 15 required
lto-wrapper.exe: fatal error: D:\Users\Zohar\Docs\Arduino\Shared_Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc returned 1 exit status
compilation terminated.
d:/users/zohar/docs/arduino/shared_arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed
collect2.exe: error: ld returned 1 exit status
Using library OneWire at version 2.3.8 in folder: D:\Users\Zohar\Docs\Arduino\libraries\OneWire 
Using library DallasTemperature at version 4.0.3 in folder: D:\Users\Zohar\Docs\Arduino\libraries\DallasTemperature 
Using library SPI at version 1.0 in folder: D:\Users\Zohar\Docs\Arduino\Shared_Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.5\libraries\SPI 
Using library mcp_can at version 1.5.1 in folder: D:\Users\Zohar\Docs\Arduino\libraries\mcp_can 
Using library EEPROM at version 2.1 in folder: D:\Users\Zohar\Docs\Arduino\Shared_Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.5\libraries\EEPROM 
Using library U8g2 at version 2.35.30 in folder: D:\Users\Zohar\Docs\Arduino\libraries\U8g2 
Using library Wire at version 1.0 in folder: D:\Users\Zohar\Docs\Arduino\Shared_Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.5\libraries\Wire 
exit status 1

```

MCUdude, 2025-08-09:
Does your code have inline assembly?

ZoharML, 2025-08-09:
Not that I know about.
It have allot of code that I wrote, none in assembly, and some libreries that I use (OLED, CAN, etc).

In the past I, changed in the setup() made this appear.
I just thought that I would make a test with moving all the setup code to the loop() + a if to limit it's run to only once, and see if it changes anything.

MCUdude, 2026-10-13:
I'm closing this as I'm not able to reproduce the issue, and it has not been reported by other users.

closed 2026-10-13 by MCUdude